trinamic/TRAMS #2: Endstop not recognized

kri5, 2016-02-26:
After wiring following the documentation,
the X endstop stays always in triggered state, while Y and Z stays always opened,
whether i trigger or not the sensor, or worse, stays in the same state even with no sensor plugged.

Any ideas?

trinamiclj, 2016-02-26:
kri5,

Will check what could be the reason. Can you share the type of sensors you are using?

Best regards
LJ

kri5, 2016-02-26:
@trinamiclj I am using some mechanical endstop such as these http://goo.gl/GQIg96 for X and Y, and a capacitive one for the Z axis (which is 12v so i wired the + on one of the VM pins).

Everything was working fine on my RAMPS board.

kri5, 2016-02-26:
Btw i have the hardware version 1.0, with the firmware 0.93 with some modification to the configuration.h to match my printer specification (corexy printer)

kri5, 2016-02-29:
I do not know if it's useful, but when not under power supply (12v unplugged but arduino mega powered on), the x endstop is recognized as open (but still does not trigger)

kri5, 2016-03-04:
@trinamiclj Any ideas?

trinamiclj, 2016-03-23:
kri5, sorry for the delay. Please note that the endstop inputs do only support +5V. The 12V sensor might destroy the input if the signal is +12V as well. If the output of the sensor is +5V it should be fine.

Another thing to clarify: How did you connect the sensors. By default only the left endstop inputs are functional. Furthermore the FW supports only one end stop per axis. Means: You have to chose either the left endstop input or the right. Both together do not work right now. If you connected you endstops on the right endstop input please either change it to the left inputs by connecting them to the left input pins or you can switch it by SW/FW. You will find a define in the "Configuration.h". The block looks like this:

//TRINAMIC TRAMS Endstop settings
//Only use ONE #define per axis
#define X_SW_LEFT
//#define X_SW_RIGHT
#define Y_SW_LEFT
//#define Y_SW_RIGHT
#define Z_SW_LEFT
//#define Z_SW_RIGHT

Here you can comment the "#define X_SW_LEFT" and uncomment the X_SW_RIGHT. Once again, you may only use one endstop per axis for now.

Does this help?

BR,
LJ

trinamic-ed, 2016-06-15:
Issue can be closed, because new firmware fixes the bug.

trinamic-ed, 2016-06-15:
Fixes #2

closed 2016-06-27 by trinamic