trinamic/TRAMS #5: Endstops still not recognized?

eraco, 2016-09-29:
I'm using the Makerbot Mech Endstops 1.8 and the m119 gcode shows no reaction when the button is pressed or even conected for that matter. i also find it weird that the x axis endstop is the only one that says triggered when connected to (24v) wall power. i even took a switch and manually soldered a pullup resistor to it and it still doesnt work, neither on the left nor the right input. am i missing something here? im running V1.00 firmware with no changes to the configs, compiled with the arduino ide 1.0.6 and 1.6.12

trinamic, 2016-10-14:
Hello @eraco,
The M119 command will not work at this point, because the endstops are connected through the TRAMS board to the TMC5130 chips and not to Arduino. They perform autonomously the homing (that is the final purpose of the endstops). The M119 code still tries to read the state of the Arduino pin, that's why the value doesn't change. This part should be updated to read the endstop state from TMC5130 (to be done).

If you are interested in the M119 command to work, we would appreciate if you could contribute to this part. I could guide you to implement how to read it via SPI from the TMC5130 chips and if you can submit your improvement to the TRAMS-Firmware repository. It would help the TRAMS community.

But anyway, the most important thing is that the homing works. The configuration of the endstops can be selected in the file TMC_TRAMS_CONFIGURATION.h. (fw v1.0). Then just try to do the homing with Repetier Host, for example.

Rob1212, 2016-10-20:
I'm having trouble with switches as well. I'm using a Microswitch like this:
http://robotechshop.com/wp-content/uploads/2015/12/Micro-Switch.jpg

It uses 2 wires: when the switch is pressed the circuit is open, when the switch is not pressed the circuit is closed. I've tried all combinations of pins:
Left+5V
Left+Gnd

I've tried all combinations in the firmware (TMC_TRAMS_CONFIGURATION.h) as well, but nothing seems to work. What type of switches are we supposed to use?

Rob1212, 2016-10-20:
Another thing is when I press the home button it always tries homing to 0mm. How do I change it to home in the other direction?

(Posted 5 min after my last message, so no the switches still don't work)

eraco, 2016-10-20:
Hi, 

my issue was that stallguard was enabled by default. The Firmware seems to use this as an endstop. Thas pretty cool but should definetly not be enabled by default. If for example Leadscrew is used or on a core xy printer the "load" isnt high enough to trigger stallguard (sensitivity could be configured but i would rather just use mech endstops).

@Rob1212 You can solder a pullup-resistor to your Switch and use it like [this](https://learn.sparkfun.com/tutorials/pull-up-resistors/what-is-a-pull-up-resistor)
or your could just use the common [mech endstops ](http://www.banggood.com/3Pcs-RAMPS-1_4-Endstop-Switch-For-RepRap-Mendel-3D-Printer-With-70cm-Cable-p-1054557.html?rmmds=search) and maybe solder your Switches to the board if you need to use this particular model.

@trinamic Now that my endstops are working, the hotend kind of backs off from the Switch and doesnt stay in the Position where the endstop is triggered. is this behavior configurable? Since the Z-Axis starts homing by first going up for a few centimeters before actualy homing and i fear it might crash into the hotend.

eraco, 2016-10-20:
@Rob1212 from the readme: 

> In configuration.h you can now choose the homing direction and the homing switch (left/right)

eraco, 2016-11-07:
@trinamic i just had a look at the Code and it seems like support for homing to max was dropped in fw1.0? i tried to reimplement it by using fw0.93 as a starting point but ran into all sorts of issues (all axis ran REALY slow after homing or not recognizing the endstop anymore). can you offer some Information on this? the tmc5130 datasheet was bit to complex for someone who does this as a hobby.

Gau54, 2016-12-04:
My homing works! with NO switch in all axes, in fw1.0. on Anet A8 printer,  like the original board. For non-virtual in X/Y axes, change the sw_mode, and pinout connect. It's good board, but my transitor for bed in 24v is shorted! my changelog:

ChangeLog For TRiNAMICS Trams V1.Firmware with Anet A8

•	HOMING
#define min_software_endstops to false (config.h 329)
set travel limits after homing for you printer model (config.h 332) 220/0 220/0 200/0
#define HOMING_FEEDRATE {3500, 3500, 3000, 0} (config.h 473)
set retract after homing hit (config_adv.h 329) 31 10 0,5
set homing procedure X first,Y, and Z (config_adv.h 1399) invert X/Y bloc-code
reducte Z climb in Z homing procedure (TMC5130.cpp 189) delay 1000
switch left/high for all (TMC_TRAMS_Configuration.h 38)
Sw_mode (400=virtualX/Y to 0x053) (TMC5130.cpp 201) with NO switchs between VCC-lelt pin for X/Z, but Y axe are VCC-rigth pin, _//no pullup-resistor //i do not know why!

•	MOTOR
current motor to 12/8 (TMC_TRAMS_Configuration.h 18) =control heat!
stepper direction X=inverse Y=normal Z=inverse Eo=Inverse (TMC_TRAMS_Confi/.h 72)

•	PID hotend/hotbed
redoo autoPID : 	-hotend M303 s200 Eo C8 (Configuration.h /the active in 158)
			        -hotbed M303 s50 E-1 C8 (Configuration.h 197)

for the moment:
#define DEFAULT_AXIS_STEPS_PER_UNIT   {100,100,400,107.14}
#define DEFAULT_MAX_FEEDRATE          {120, 120, 8, 50}
#define DEFAULT_MAX_ACCELERATION      {10000,10000,3000,3000}
#define DEFAULT_ACCELERATION          8000    
#define DEFAULT_RETRACT_ACCELERATION  8000
#define DEFAULT_XYJERK                10    
#define DEFAULT_ZJERK                 0     
#define DEFAULT_EJERK                 5.0

closed 2017-04-11 by eraco