trinamic/TRAMS #6: Motor header connectors

Rob1212, 2016-10-20:
I've been having all sorts of trouble with my TRAMS board, at one point I thought I had a defective unit. It turns out my motors are coming loose from these half height header pins you guys decided to use.

How about some part numbers for the correct female locking connectors? Part numbers from Digikey would be nice.

eraco, 2016-10-20:
The Connectors are "JST EH"-Series. You can find the housings [here](http://www.digikey.de/short/391q8d) and the crimping connector [here](http://www.digikey.de/product-detail/en/jst-sales-america-inc/SEH-001T-P0.6/455-1042-2-ND/527267). But good luck finding a suitable crimping tool. If you are based in Germany i highly recomend getting this [crimping pliers insert](http://nessel-elektronik.net/Crimp-Einsatz-C-fuer-Crimpzange-N-EH-XH-Kontake). Otherwise i have heard good things about the "Engineer PA-09"

Rob1212, 2016-10-20:
Eraco, thank you for all the help today. I was worried I wouldn't be able to find these connectors. For anyone else reading this and getting ready to order parts they might also be missing, be sure to order some fuses as well. I bought my TRAMS from Digikey and it came with two 5 Amp fuses, instead of one 5 Amp and one 10 Amp. Someone else on here mentioned they blew their 5 Amp when they turned on their heated bad, so I will be ordering some of these as well:

http://www.digikey.com/product-search/en/circuit-protection/fuses/655421?k=TR5+373+SERIES&k=&pkeyword=TR5+373+SERIES&pv66=25&pv66=1&FV=fff4000a%2Cfff8003d%2C1080001%2C1080019%2C108001b%2C108056e&mnonly=0&newproducts=0&ColumnSort=0&page=1&stock=1&quantity=0&ptm=0&fid=0&pageSize=500