trinamic/TRAMS #7: Merging TRAMS-specific code back into Marlin

hoppersoft, 2017-03-16:
Hello! I'd like to try out a TRAMS board, but I want to keep some of the newer Marlin features such as bilinear auto-leveling. Is there any way to see at what commit the 1.0 TRAMS firmware was forked from the Marlin repo? I'd like to see if it would be possible to perform a merge and bring everything up to date with the latest Marlin codebase.

trinamic, 2017-03-22:
Hello Andy,
really good timing for this issue, since we have the intention of merging the TRAMS Firmware back in Marlin. We are considering the options we have.
I tried to trace back the commit from which we forked Marlin, which isn't easy. We did it local, not in github. The oldest code I've found is from July 29th 2015. I could send you the zip if needed.
However, not only Marlin has evolved, but our changes are quite deep in the firmware. In the attached document we explain the conceptual differences and I'm currently writing a document about how to make the firmware adaptation for a TRAMS-like firmware (the diagram belongs to this new one).
I think the best option, and what we wanted to do, is to fork current Marlin and try to implement our changes there. We would really appreciate the collaboration.
Sorry for the long text. What do you think?
[AN035-CNC_motion_with_ramp_generator.pdf](https://github.com/trinamic/TRAMS/files/862043/AN035-CNC_motion_with_ramp_generator.pdf)
[image]

hoppersoft, 2017-03-22:
I'd love to help! If you send me the zip, I'll take a stab at the code
forensics to see if we can't nail down a commit hash where it diverges.
From that, we could generate a diff from that commit and see just how hairy
it would be to attempt replicating those changes against the latest
RC/RCBugFix branch.

I was looking into ordering a TRAMS board, so this may be my motivation to
pull the trigger!

<https://cloud.githubusercontent.com/assets/1303508/24207703/d9e2a75a-0f21-11e7-9347-200247994393.png>

<https://github.com/trinamic/TRAMS/issues/7#issuecomment-288449874>, or mute
the thread

trinamic, 2017-03-23:
Hi, here is the .zip.
[Original_Marlin_for_TRAMS.zip](https://github.com/trinamic/TRAMS/files/864901/Original_Marlin_for_TRAMS.zip)
The main changes are: the interrupt routine in stepper.cpp, the homing function in Marlin_marin.cpp. We added also an intermediate queue and a pre-calculation function in stepper.cpp, SPI communication, and some TMC5130 own functions. Among other minor changes like st_synchronize() and so.

jsidebottom, 2017-05-29:
The contents of Original_Marlin_for_TRAMS.zip is identical to Marlin_TRAMS_V0.9.zip in your repo.
Judging only by the unmodified files in that zip, the last commit before you started to diverge from https://github.com/MarlinFirmware/Marlin was:

Revision: ae12ad0d6c4ecd7f0d5e476a1267034c4467640e
Author: Alexander Hirzel <alex@hirzel.us>
Date: 2/01/2015 5:36:27 PM
Message:
Fix typo, PID_BED_POWER -> MAX_BED_POWER

trinamic, 2017-06-02:
Thanks! For your info, @teemuatlut has started to merge the TRAMS Firmware into Marlin and has progressed quite a lot. He's working on [this](https://github.com/teemuatlut/Marlin/tree/trams-updated) branch of his fork

hoppersoft, 2017-06-02:
I'm terribly sorry; life intruded and I never found the time to get that
going. I noticed the flow of TRAMS commits!

jsidebottom, 2017-06-02:
I hadn't noticed that branch, looking at those commits now. Thanks for linking it.

trinamic, 2017-06-07:
@hoppersoft Nevermind!
@jsidebottom You're welcome. Looks promising!