trinamic/TRAMS #9: Implementing homing to max

eraco, 2017-04-11:
Homing to max doesn't seem to be implemented currently. Any suggestions on what im doing wrong? My current TMC5130.ccp: https://gist.github.com/eraco/f22a37971532f776e0843b4af704ecdc .

trinamic, 2017-04-24:
Hi @eraco, you mean homing when the endstop is not at 0 but at the other side, right?

I don't see any problem in your changes so far. Can you describe the problem?

You could use SWITCH_POSITION_X, SWITCH_POLARITY_X, STEPPER_DIRECTION_X to determine X_HOME_DIR

eraco, 2017-11-08:
Hi @trinamic ,

so i decided to give the Trams another shot. I followed your suggestion and tinkered with the variables. SWITCH_POLARITY_X and STEPPER_DIRECTION_X seem to work fine as far as i can tell. But switching SWITCH_POSITION_X from left to right and vice versa just changes the pin but not the homing direction. 
The TMC5130 Datasheet states: "Move to a more negative position for a left switch, to a more positive position for a right switch". For my Ultimaker2 this would mean X left and Y,Z right. (from UM2 configuration.h: #define X_HOME_DIR -1, #define Y_HOME_DIR 1, #define Z_HOME_DIR 1)

If i am not mistaken the culprit is that rampmode is always set to VELOCITY_MODE_NEG before homing. If i try to set it to VELOCITY_MODE_POS for the corresponding axis it moves towards the endstop but doesnt trigger it. (Stallguard is off) When im homing with VELOCITY_MODE_NEG the switch works as intended.

Also XTarget and XActual are always set to 0 after homing. If im homing to max this should be set to the lenght of axis in steps, right?

And one last question: What does X_CURRENT_RUN define? Is it the amps that go to the motors? How is it calculated? Mine are rated for just 1 amp. I know coolstep adjusts this automagicaly but i would like to cap it at 1amp, just to be safe.

I look forward to hearing from you.

eraco, 2017-11-09:
Okay seems like you tried doing somthing along the lines of what i posted earlier (from fw0.93):

			if(X_HOME_DIR < 1)
			{
				VELOCITY_MODE_POS = 0x02;
				VELOCITY_MODE_NEG = 0x01;
			}
			else
			{
				VELOCITY_MODE_POS = 0x01;
				VELOCITY_MODE_NEG = 0x02;

			}
			#ifdef X_SW_LEFT
						SW_CONFIG = 0x25;
			#endif

			#ifdef X_SW_RIGHT
						SW_CONFIG = 0x35;
			#endif
			break;

I noticed that you used other values for SW_CONFIG than what the bitwise OR currently returns. i did some reading in the datasheet on the SW_MODE Register but could'nt make much sense out of it. Could you provide some explanation on how to use it?