trinamic/TRAMS #10: 5V, schottky diodes, and other wishes

swe3tdave, 2017-05-17:
I am a huge newbie/hobbyist but there are a few things that bugs me after looking at the TRAMS schematic and eagle files...

Am i reading this wrong or the 5V regulator from the tmc5130 is in use? The shield/arduino doesn't seem to be hooked up to the VCC of the tmc5130, but the resistor seems to indicate an external power supply?

It would be nice to be able to hook up an external 5v power supply, like an atx psu. i dont see a PS_ON header, i think its a bit sad, but i can live with it... :)

The tmc5130 seems unable to support more than 1.2A without schottky diodes, so why isn't there a way to add some? i don't even care if they're not included and i'm fine hooking up the second lead to the motor cable directly.

There seems to be traces missing for the 5V going to the servo pins and the endstop connector. Why even use 5 pin connectors if the 5V is not hooked up?

There are only one set of pospulse/interrupt on the shield, so they seem to be hooked up together, i supposed these are for the stallguard. But i was asking myself, what if someone wants to use the TRAMS for offsetting calculation from their arduino mega and actually use a bigger stepper? Right now they could not and i think its a bug. i wonder if you could use the mega for delta calculation and the drivers for step generation? I'm done trying to make a delta print correctly but someone might like the idea...

Last question, i was interested by the possibility of using FULL STEP and having 256 microstep generated, does the hardware support that option? I understand that the software doesn't but it seems pretty easy to modify. I want to have the precision of FULL STEP with the noise cancelling of the microstep, is that possible? Am i too greedy/paranoid about this? I am transfering from delta to Cartesian printer and the thought of not having 0.05mm precision/repeatability really bugs me. If i print a square it needs to be square!

swe3tdave, 2017-05-17:
ok i get the 5v regulator thing, that's the standard application circuit... if i understand, at 12v the dissipation of the regulator should be 37% less than with full power... that's beautiful and all but i've burned enough drivers to be more than a little paranoid about power dissipation... i wonder, the motor generate electricity when they turn with no power applied, is that enough to burn the driver? is that what the page 20 in the tmc5130a datasheet is about?

trinamic, 2017-05-22:
Hi @swe3tdave, thank you for your message. I've tried to divide it into parts, see the answers inline.

> Am i reading this wrong or the 5V regulator from the tmc5130 is in use?

Yes, the internal 5V regulator is used for driving the internal logic. For the I/O the external 5V from the Arduino is used. That is also the reason why you must power on the PSU (VM) before the external 5V. (see datasheet p. 12 under point VCC). The schematic used in the TRAMS design is taken from the datasheet example on p. 13 and 121 and it's the same as in the TMC5130 Evalboard.

> The shield/arduino doesn't seem to be hooked up to the VCC of the tmc5130, but the resistor seems to indicate an external power supply?

The Arduino powers the I/O Pins. The VCC Pin of the TMC5130 is driven by the TMC5130 5VOUT pin through a 2.2 Ohms Resistor. The reason is found in the datasheet on p. 120 under Supply Filtering.
 
> It would be nice to be able to hook up an external 5v power supply, like an atx psu. i dont see a PS_ON header, i think its a bit sad, but i can live with it... :)

Theoretically, you could use one of the unused pins in AUX-1 or AUX-2 to switch an external PSU on or off. These are pin compatible to the original RAMPS board

> The tmc5130 seems unable to support more than 1.2A without schottky diodes, so why isn't there a way to add some? i don't even care if they're not included and i'm fine hooking up the second lead to the motor cable directly.
        
While developing and testing the TRAMS there was no need for higher currents. We tested the TRAMS with standard NEMA17 motors on a Prusa I3. Unfortunately, the only way would be to solder a wire directly onto the sense resistor pads (see Image)        
[image]

> There seems to be traces missing for the 5V going to the servo pins and the endstop connector. Why even use 5 pin connectors if the 5V is not hooked up?
        
The servo and the endstop connectors are both correctly connected to +5V.   
     
> There are only one set of pospulse/interrupt on the shield, so they seem to be hooked up together, i supposed these are for the stallguard. But i was asking myself, what if someone wants to use the [...]
        
I do not understand what you mean by offsetting calculation from their Arduino and using a bigger stepper, especially in the context of the pospulse/interrupt.        
 
> i wonder if you could use the mega for delta calculation and the drivers for step generation? I'm done trying to make a delta print correctly but someone might like the idea...
    
Because the Arduino does not generate any steps (see next answer) it should be easy to just use your calculated delta distances instead of the cartesian distances. 
 
> Last question, i was interested by the possibility of using FULL STEP and  having 256 microstep generated, does the hardware support that option? I understand that the software doesn't but it seems [...]
        
The key difference between the TRAMS hard/software to the "standard" (RAMPS, MELZI, etc.) is, that the Arduino does not produce any steps at all. It only calculates the distance, velocity and acceleration for each axis then sends it to the TMC5130 via SPI. The TMC5130 then calculates and generates the steps internally. So there is no interpolation from 1/16 steps like with the TMC2100 on the SilentStepStick ) up to 1/256 steps. The TMC5130 directly calculates in 1/256 steps.
This way you get a very high resolution and a silent printer. You could also activate the stealthChop™ feature on the TMC5130 to make it even quieter. Although some people claim that stealthChop™ leads to lost steps etc. we found the printer to be as reliable as without it.
 
> ok i think i get the 5v regulator thing. its the standard application circuit and the current is too low too bother adding components to prevent heat there...  i wonder, the motor generate electricity when they turn with no power applied, is that enough to burn the driver? is that what the page 20 in the tmc5130a datasheet is about?
 
Theoretically, it would be possible to destroy the drivers with high voltage peaks, but for that, you would have to turn the stepper motors really fast and in a "normal" 3D printer that case is very unlikely.