trinamic/TRAMS #12: Endstops not working even after adding external 10k pullup

pdeclerc, 2017-09-30:
Hi,
I'm trying to get a "non functioning" pursa style 3d printer running with the TRAMS.
I've connected the existing endstops (for home position) but they are not responding. These are microswitches.
I've added a 10k pullup between the "left" pin (microswitch contact) and the 5V and the other contact to the GND. By default the printer has A NC contact for the endstops, but even modifying this to NO does not solve the issue. The stepper motors keep on turning after pressing the endswitch.

As I understand the motor control circuits used on these boards are managing the steps and so probably also the endstops.
I've tried to modify the behaviour from the arduino code (configuration.h) but whatever I change, the axis (using the x-axis to test) is not stopping when I press the "home" button (for this specific axis) in repetier.
I'm using trams V1.0.0 and Repetier host v2.0.5.
Yes I saw the previous (closed) issue regarding the endstops. I tried everything noted there (and even more). My multimeter is saying 4.5V on the "left" pin when the switch is released and 0V when the switch is engaged (in NO config)

pdeclerc, 2017-10-31:
I tried to contact the Trinamic support, and a sales rep from Trinamic, but got no answer to my questions :-(.

eraco, 2017-11-08:
Have you tried disabling #define STALLGUARD_X / #define STALLGUARD_Y in TMC_TRAMS_CONFIGURATION.h?